OpenCyphal/specification #1: Legal statement

pavel-kirienko, 2018-07-09:
Currently, the legal statement says this, which is not exactly correct:

> UAVCAN is an interface standard open to everyone. No copyrights are reserved and no licensing is necessary for its implementation, distribution, or use. In no event shall the authors of the standard be liable for any damage arising, directly or indirectly, from its use.

A better wording should be found. The objectives are the following:

* We want the spec and related docs to be free and open in the spirit of [CC-BY-4.0](https://creativecommons.org/licenses/by/4.0/) except that the "No additional restrictions" limitation should be lifted.
* The liability disclaimer should be there.

pavel-kirienko, 2018-08-21:
@thirtytwobits Can your [*OSS guru*](https://github.com/UAVCAN/pyuavcan/pull/38#issuecomment-406670151) be of any help here?

thirtytwobits, 2018-08-31:
Unfortunately not. Sorry. I reached out but nobody had advice on this beyond pointing out that AWS uses CC-BY-SA-4.0 on their github documentation. Not sure if that's helpful here.

Perhaps we should sniff around other open standards and see what they have. Here's my quick search:

https://xmpp.org/extensions/xep-0009.html#appendix-legal
https://www.xiph.org/ogg/doc/rfc3533.txt
https://www.w3.org/Consortium/Legal/2015/doc-license

PR #18 (2018-09-13): Fixed the legal statement
Using CC BY 4.0 as advised.

Fixes #1.

[image]
closed 2018-09-13 by pavel-kirienko