OpenCyphal/specification #2: Protocol update effort (tracking issue)

kjetilkjeka, 2018-07-17:
This is an attempt of gathering references to everything concerned by the new protocol. The list will be edited to be kept updated. To make this issue helps instead of just becoming yet another place to discuss things, please keep specialized discussion in the relevant issue.

 - [ ] Settle on how to do versioning and stabilization
     - https://github.com/UAVCAN/uavcan.github.io/issues/39
 - [ ] Decide what to do with TAO
     - The current discussion lean towards not using TAO for the new protocol, only CAN-FD is decided not to use TAO for. (https://github.com/UAVCAN/uavcan.github.io/issues/34)
 - [ ] Decide how ID fields should look like
     - Need to settle on one of two alternatives in (https://github.com/UAVCAN/dsdl/issues/35#issuecomment-405544765)
 - [ ] Decide how dsdl version info should be conveyed over the wire
    - The discussion is currently leaning towards putting it in the ID field (https://github.com/UAVCAN/dsdl/issues/35#issuecomment-405544765)
 - [ ] Decide where the tail byte should be put in the existence of padding
     - The tail byte should always be put on the DLC boundary (https://github.com/UAVCAN/uavcan.github.io/issues/35)
 - [ ] Decide where CRC should be put in the existence of padding
     - The CRC should be put directly before the tail-byte and directly after the padding bytes (https://github.com/UAVCAN/uavcan.github.io/issues/36)
 - [ ] Settle on what CRC should be initialized with
     - The discussion leans towards not having compatibility information in the CRC

kjetilkjeka, 2018-07-17:
@pavel-kirienko feel free to add/edit stuff, or uncheck things if i were to quick to check them.

pavel-kirienko, 2018-07-19:
I marked the TAO entry as decided since the consensus seems to be in favor of its complete removal. Bye bye TAO.

kjetilkjeka, 2018-08-03:
> Settle on whether CRC should include data type signature (or other forms of compatibility analyzis).

I think we've practically reached consensus on that the CRC should not include data type signatures or similar. Check off your ticky box if you agree to the conclusion and that the discussion has been left open long enough. Or edit and add another ticky box if you want to bring the attention to someone else.

 - [ ] @kjetilkjeka 
 - [ ] @thirtytwobits 
 - [ ] @pavel-kirienko

pavel-kirienko, 2018-08-11:
> It has been suggested to use a bit in the transfer ID, but the current discussion is leaning towards not conveying protocol version info over the wire.

I think the discussion is leaning towards using the toggle bit for that. I took the liberty to reflect that in the draft (sec 4.2.3.2), believing that that was the consensus:

> Transfers where the initial value of the toggle bit is zero must be ignored. The initial state of the toggle bit may be inverted in the future revisions of the protocol to facilitate automatic protocol version detection.

kjetilkjeka, 2018-08-14:
Yes you're absolutely right. Thanks for pointing this out, I've updated the overview now.

pavel-kirienko, 2018-12-11:
Marked the following as resolved:

- Settle on how the protocol version should be conveyed over the wire
- Figure out how we handle deprecations of definitions.
- Figure out if we should do anything to namespacing of dsdl

pavel-kirienko, 2018-12-14:
All important things have been settled upon and resolved, so I am closing this. For future reference, many important decisions have been discussed there: https://forum.uavcan.org/c/dev

closed 2018-12-14 by pavel-kirienko