OpenCyphal/specification #4: Proposed New High-Level Definition of UAVCAN

thirtytwobits, 2018-07-18:
[image]

Changes:

1. Core data types are distinct and required.
2. Standard data types (e.g. equipment) are distinct and optional.
3. Custom data types are first class members of the specification.
4. The interface between the serialization of data types and the transport layer is well-defined.

Item 4 was controversial from the google group start of this proposal but I still think there is value to allowing for a different data serialization layer to sit on top of some UAVCAN implementations (e.g. using libuavcan but not DSDL). I'd like to at least propose it for the record here in our specification repository.

pavel-kirienko, 2018-07-18:
"Standard functions" in the context of the specification refers to the high-level features of the protocol. As such, this block should rest on top of "Core data types". Unless I'm misreading your proposal.

I am considering to decimate the set of standard data types from the `uavcan.equipment` namespace; right now it seems to be poorly organized and some of the definitions might be overly application-specific. I think we should postpone this until the transport layer is settled and agreed upon.

thirtytwobits, 2018-07-18:
Updated the diagram to put the Standard Functions above the Core data types

pavel-kirienko, 2018-08-03:
I suggest closing this as the relevant suggestions have already been incorporated into the current specification draft.

thirtytwobits, 2018-08-07:
I still don't like your diagram Pavel ;-) Can we keep this open until we finish debating the proposal I put in my latest specification review around three classes for data types; required, optional, and standard?

PR #23 (2018-09-19): Issue #16 edits to section 2.1
This includes a proposal to close #4  if we can agree on this diagram.
closed 2018-09-19 by pavel-kirienko