OpenCyphal/specification #5: Rigorous definition of semantic compatibility and bit compatibility

pavel-kirienko, 2018-07-19:
So I was writing the new spec doc and got stuck at the semantic and binary compatibility part. I'm leaving the lab right now with the intention of getting back to this first thing in the morning. @kjetilkjeka if you happened to have good definitions at hand, please share here.

This is what I ended up drafting so far:

```latex
\section{Data type compatibility and versioning}\label{sec:dsdl_versioning}

\subsection{Rationale}

As can be seen from the preceding sections,
the concept of \emph{data type} is a cornerstone feature of UAVCAN,
which sets it apart from many competing solutions.

In order to be able to interoperate successfully,
all nodes connected to the same bus must use compatible definitions of all employed data types.
This section is dedicated to the concepts of \emph{data type compatibility}
and \emph{data type versioning}.

A \emph{data type} is a named set of data structures defined in DSDL.
As has been explained above, in the case of message data types,
the set consists of just one data structure, whereas in the case of service data types
the set is a pair of request and response data structures.

Data type definitions may evolve over time as they are refined to better address the needs of their applications.
In order to formalize the data type evolution process with respect to the data type compatibility concerns,
UAVCAN introduces two concepts: \emph{semantic compatibility} and \emph{binary compatibility},
which are discussed below.

\subsection{Semantic compatibility}

\subsection{Binary compatibility}
```

kjetilkjeka, 2018-07-19:
I think that compatibility is about the set of features being a subset/superset of some other set of features.

I think binary compatibility is the least difficult. Perhaps binary compatibility can be defined as something in the lines of the following.

We say that a definition (the compatible definition) is *binary backwards compatible* with another definition (the original defintion) if all the following statements hold:
 - Every non void field in the original data type must also exist for the compatible data type. (A field with the exact same name, type and position in regards to serialization must exist).
- For every non-void field that exists in the compatible definition that do not exist in the original definition there must be a void field in the original definition encapsulating (in regards to serialization) the whole new non-void field.

Semantic compatibility is more tricky. I have summed up a few properties i think we need to have without attempting to write a formal definition.
 - When adding a new const, there must be a sensible default that the old definition may use (For instance adding new error types is Ok, since every value besides the `OK` value must be treated as an unknown error)
 - When adding new fields the void field serialization `0` must make sense. 
     - For a status field `0` must be the equivalent to "nothing to report". 
     - For a value field there are several hacks. 0 can mean "no value" or if the value `0` is needed one of the bits of the void field can be used as a "value exists" bit.
 - When adding a new field existing fields must not change their behaviour based on values from the new field. However, the new field may change behaviour based on existing fields.

And at last: 
 - "We say that a definition is *backwards compatible* with another definition if it is both semantic and binary compatible with this definition.
 - We say that a definition is simply *backwards compatible* If it is backwards compatible with all definitions sharing the same major version number and have smaller minor version number"

kjetilkjeka, 2018-07-21:
Ok, so binary compatibility is only about serialization. That makes sense. My definition mixed in what applications can expect as well. My proposed constraints will implicitly enforce both the bit compatibility constraints and some other constraints. But it may still be valuable to have a clear split between them.

Your example is then indeed binary compatible. But if one attempted to update the definition in an application the new code would not compile. And due to that I would say that definition B should not be considered compatible (in the sense of "total compatibility") with definition A. I see two simple ways to keep them compatible. 
 1. Instead of changing types introduce bitmasks `uint64 STATUS_FLAG_BM = 0xffffffff00000000`
 2. Introduce a concept of untagged unions in dsdl (this might very well be undesired as it makes things a bit more weird and may be difficult to support for some languages).

I think I described the least strict sensible definition for bit/binary/serialization compatibility in https://github.com/UAVCAN/uavcan.github.io/issues/40 there's no good precise definition there but it can perhaps be used for inspiration? To sum it up, all it cares about are the layout of static fields vs dynamic fields vs union fields.

pavel-kirienko, 2018-07-22:
Closing for #6

closed 2018-07-22 by pavel-kirienko