OpenCyphal/specification #6: Review of the chapters 1, 2, and 3

pavel-kirienko, 2018-07-22:
Here comes an early draft for an early review. This PDF is compiled directly from the current master: [UAVCAN_Specification.pdf](https://github.com/UAVCAN/specification/files/2217431/UAVCAN_Specification.pdf)

I would like @kjetilkjeka and @thirtytwobits to have a critical look. I am going to review everything once again when the content is finished to fix the remaining styling and wording issues.

The DSDL chapter is rather different from what is published on the website right now: TAO is gone, and the serialization section is moved up.

pavel-kirienko, 2018-07-23:
Updated [UAVCAN_Specification.pdf](https://github.com/UAVCAN/specification/files/2220864/UAVCAN_Specification.pdf)

scottdarch, 2018-07-24:
[UAVCAN_Specification_thirtytwo.pdf](https://github.com/UAVCAN/specification/files/2221806/UAVCAN_Specification_thirtytwo.pdf)

Comments are in the document. Let me know if this review process works.

I stopped at chapter 3 because I didn't realize that the datatype identifiers for top-level types were optional and this has me really confused. How do you envision such a feature is used? If a type id is determined at runtime then which node determines it and how do the other nodes know what typeid was chosen? I feel like I'm missing something big here because I can't envision how this could possibly work.

pavel-kirienko, 2018-07-25:
@scottdarch let's move to https://github.com/UAVCAN/specification/pull/7

> Comments are in the document. Let me know if this review process works.

It does.

kjetilkjeka, 2018-07-29:
I'm sorry this took so long @pavel-kirienko I've been away the last week and will be away until Wednesday. I managed to write some comments from the last draft you linked. I will be able to review/discuss more on Thursday. 

uavcan review:
- 1.2 is makes me belive that datatypes can be changed without incrementing minor version. It should be changed to make it clear that datatypes are immutable. Section 2. explains this well.
- There are DSDL definitions that break the nice layering. The most important is the NodeStatus message. The protocol is in fact dependent on this definition and it can not be updated without breaking the protocol. It should be mentioned that the NodeStatus message is special somwhere (towards the end perhaps) of section 2.
- What is the status of changing up namespaces to reflect functions/domain (aero, ground, etc)? This is not done in your draft. We should probably create an issue or similar to discuss this, right?
 - 3.6.2.2 are unions fixed or variable sized data structures. I think dynamic? In that case, there's an error in the text.
 - 3.6.2.1 Valid encoded representation needs defining. One of the things im unsure of: Is a void field encoded to a non-zero value valid encoded?
 - 3.6.3.2 If we allow fields to change name we will break code by incrementing minor version. I'm not sure if this is something we want to allow. Especially since there often exists none breaking ways to work around it. I think we should create an issue and discuss this.
 - 4.1.1.2 Perhaps mention here or later that when creating a datatype that is intended to be used for anonymous frame transfer it must fit inside a single frame transfer for the transport protocol it is intended to be used with.
 - 4.1.2.1 I'm not sure the "• The server should be able to process any request in under 0.5 seconds." requirement makes sense to impose that strongly. Should perhaps be replaced with something more general like "If the server uses a significant part of the timeout period to process the request, the client might drop the request before receiving the response. It's good practice to make sure that the server will be able to process any request in less than 0.5 seconds."
 - 4.1.3 I'm not sure that these mnemonics makes sense. In my mental model there are really only two type of priority levels. Realtime and non-realtime (or background if you want). All real-time transfers needs to finish before it's deadline. And the way to achieve this is by giving high priority to messages with close deadlines, not by some importance value. Words like "Emergency" and "Critical" is thus misleading. Another problem is that these priorities only make sense relative to other and need some sort of globabl analysis to make sense. Choosing priorities is a difficult subject that is very often done poorly. I propose the following things that might be helpfull:
     1. In this section we should explain the considerations needed to make when choosing priorities
     2. We need to choose mnemonics related to urgency instead of importance. Starting with "Background" as the lowest is nice.
     3. We could also create some ballpark guidelines of what kind of deadlines that should be associated with the different priority levels.
     4. At the end of the day setting priorities needs to be a global analysis for a system (except for cases where background will always make due). And it should be encouraged to make changing priority levels possible dynamically.

pavel-kirienko, 2018-07-30:
@kjetilkjeka Thanks! I didn't have a chance to act upon your feedback yet; I'm leaving until Thursday myself, will continue then. Meanwhile, I have finished porting the chapter 4 (CAN transport layer) to PDF; here it is attached (built from master): 
[UAVCAN_Specification.pdf](https://github.com/UAVCAN/specification/files/2242298/UAVCAN_Specification.pdf)

The last sections of the CAN transport chapter (transfer reception) feel quite haphazard; I'm going to review and possibly rewrite them later.

kjetilkjeka, 2018-08-02:
First a general comment: I think it would be better to avoid introducing can before neccessary. I would even refrain from calling the section "CAN bus transport layer", instead i would call it "The transport layer". Then first every concept of the transport frame could be defined (this is basically 4.1+4.2+4.4 just use the word transport frame instead of CAN Frame). Then the CAN transport layer could be it's own section (e.g 5) or subsection(s) (e.g 4.4). If this route is chosen, a definition similar to the following should exist - "**Transport Frame** - a frame of the underlying transport protocol (e.g CAN2.0B or CAN-FD)."

I've tried to tie these specific comments into the existing structure to make it easy to navigate.

# 4 CAN bus transport layer
 - Even though it's not incorrect to call CAN 2.0 legacy, as CAN-FD is both a better and newer version of CAN 2.0, I would like to avoid this term due to possible negative conotations when used for software. The masses are not going to make the change over night, and neither should we. I think we should show understanding of users not able to upgrade at this point by referaining from refering to it as legacy. For now, I think we should treat these protocols equal in terms of support and wording, except that we acknowledge CAN-FD to be an technical advancement.
## 4.1 The concept of transfer
 - Consider remarking that the set of single/multi -frame transfers are not equal between different transport protocols.
 - In table 4.1, row "Source Node ID" - should it say that anonymous transfers should have ID 0 instead of saying "excepting"?
 - Should message/service 
### 4.1.1
 - 4.1.1.1 Consider to tie (4.1.1) Message broadcasting closer (more clearly) together with the earlier defined concept of "message transfer".
 - 4.1.1.1 Consider changing/removing the word Regular in "Regular message broadcasting". Is "non-anonymous" meant by regular? I don't think it's safe to assume that the readers will understand that.
 - 4.1.1.1 Consider moving the (new info) fact that node IDs needs to be unique to "Basic concepts" (section 2) and only state that a Node ID is required for "regular message broadcasting".
 - 4.1.1.1 "An exception applies to anonymous message broadcasts." If regular is supposed to mean "non-anonymous" then this is contradicting as it indicateds that anonymous transfers also are regular.
 - 4.1.1.2 Consider changing "from a node that does not have a node ID" to "can be sent from a node that does not have a node ID". Since it's not prohibited for a node with Node ID to send an anonymous transfer.
 - 4.1.1.2 Should passive mode definition instead live in section 2 together with the Node ID definition introduction?
### 4.1.3
 - Look at my previous comments.
## 4.2
 - 4.2.2 and 4.2.3 - It's very suprising that single/multi frame transfers are defined here when they are introduced in 4.1, the text is so short that I believe everything relating to this concept should be put in the same place. I think the introduction should be removed from 4.1 and then the full definition could be moved directly after the message/service definitions (before transfer ID computation).
## 4.3
 - Consider phrasing the compatibility concerns more pessimistically. For instance "Even though UAVCAN currently can share the bus with other protocols based on standard (non-extended) CAN frames with 11-bit identifiers it is not guaranteed that this property will hold in the future. Doing so is not endorsed and would be done on the users own account. It would require the user to follow protocol updates closely not to break anything."
 - Table 4.6 - I find it much harder to read when vertical and not color coded. Consider using a landscape (horizontal) oriented page and/or use similar color coding as the web page.
 - 4.3.2.2 didn't we end up on recommending padding bytes to be 0x55 instead of requiring this? This is really just a nit, and I'm totally fine with requiring it if you want to. What about the padding bits? Wouldn't the same consideration apply to them?

kjetilkjeka, 2018-08-02:
I have a couple of questions regarding the redundant interfaces. 
 1. Do we have to require the users to let the first one time out before using the next one. Wouldn't an implementation where the interface where the message is received first on was used and then detection of a lost interface could be done even after the message was received on the "backup" interface.
 2. Do we need every node to use the same interface as number 1. and number 2. If we have nodes that didn't require redundancy wouldn't it be perfectly fine to split them on the two interfaces for load sharing reasons?

thirtytwobits, 2018-08-07:
[UAVCAN_Specification.pdf](https://github.com/UAVCAN/specification/files/2265040/UAVCAN_Specification.pdf)

Comments in doc.

pavel-kirienko, 2018-08-10:
Closing once again, let's discuss things at #11.

closed 2018-08-10 by pavel-kirienko