OpenCyphal/specification #9: DSDL compatibility analysis

kjetilkjeka, 2018-08-09:
I've open this issue to discuss how changes to data types should be handled. This issue is a follow up on https://github.com/UAVCAN/dsdl/issues/35, https://github.com/UAVCAN/specification/issues/5 and https://github.com/UAVCAN/specification/issues/6 but tries to isolate the compatibility discussion without blending it together with other review related items.

I will attempt to keep the text under the line updated to reflect the consensus of the discussion. The initial text is based on previous discussion and (where there are little discussion) my first instincts. Please criticize so we can improve upon it.

---

### Compatibility
Compatibility is a relationship that message definition `A` can have with message definition `B`. If `A` is compatible with `B` and `B` is compatible with `A` we say they are mutually compatible. The different forms of compatibility is defined below.

#### Bit compatibility
A structure definition `A` is bit-compatible with a structure definition `B` if any valid encoded representation of `B` is also a valid encoded representation of `A`.

#### Semantic compatibility
A data structure definition `A` is semantically compatible with a data structure definition `B` if an application that correctly uses `A` exhibits a functionally equivalent behavior to an application that correctly uses `B`, and `A` is bit-compatible with `B`.

#### Code compatibility
We say that `A` is *code compatible* with `B` if valid use of code generated from `B` also is valid use of code generated from `A` with the same spec-adhering compiler.

For a compiler to be spec-adhering it must generate compatible code if:
 - All variable names that exists in `B` must also exist in `A`. They must also have the same type.
 - All const names that exists in `B` must also exist in `A`. They must also have the same type.

#### ID compatible
We say that `A` is *ID compatible* with `B` If `B` has a default ID, and `A` has exact same default ID or `B` does not have a default ID.

### Compatibility guarantees
 - For definitions with different major version numbers, no compatibility guarantees are given.
 - For definitions with major version equal `0`, no compatibility guarantees are given.
 - For definitions with major version not equal `0` and identical path the version with the highest version number will have the following compatibilities with the other version.
     - Bit compatibility
     - Semantic compatibility
     - Code compatibility
     - ID compatible

### Mutability guarantees
 All definitions are immutable. This means that after a definition is published a new definition with the same namespace and name must have a higher version number. There are a few exceptions:
 - Whitespace formating is allowed to change on published types
 - Comments are allowed to change on published types, as long as this doesn't affect semantic compatibility.
 - Version 0.0 is not considered in any way immutable and may change in any way it might please.

### Open questions
 - How do we express deprecation?
 - Should all definitions with major version == 0 be considered mutable or only 0.0?
 - In the specification semantic compatibility implies bit compatibility. Should this be the rule, or should it be orthogonal concepts?

kjetilkjeka, 2018-08-09:
@pavel-kirienko i think the most important question is the code compatibility. You seem to opting out of it completely when writing the specification. We should discuss the value of being able to do invasive changes within one major version vs stability in code when using the same major version. 

Even though I've described the most conservative approach there exists middle grounds. For instance not guaranteeing that fields will be of the same type, but all valid const assignments before the update are also valid after the update.

The argument for doing invasive changes within one major version you presented the last time was https://github.com/UAVCAN/specification/issues/5#issuecomment-406559281 i think this problem is not very big as it can be fixed with bit masks and would probably have been fixed in the 0.X phase, and therefore not a good reason to allow breaking code compatibility. Do you have examples with more serious problems?

The reason we need (even if it turns out to be a more relaxed sort) code compatibility is that the code generated with the dsdl compiler will also need to adher to semantic versioning for the dsdl types to be truly useful in dependency systems using semver (like cargo). I think a stricter form of code compatibility will produce code that is nicer to use, so this will be a compromise.

pavel-kirienko, 2018-08-09:
> How do we express deprecation?

I have added a `@deprecated` keyword to the current draft a couple of weeks ago; it's supposed to be translated into language-specific deprecation markers, e.g. `[[deprecated]]` in C++ or `DeprecationWarning` in Python.

> Code compatibility

I think this is a bad idea. The specification should only care about bit compatibility and semantic compatibility; adding more guarantees undermines the extensibility of the standard. Yes, that implies that when an application migrates from one minor datatype version to another it may have to update the code that uses the datatype, but that is acceptable and doesn't even go against the principles of semantic versioning. What we need to be focused on is whether different nodes can successfully communicate and cooperate using data type definitions with different minor version numbers. The following example provided in the specification illustrates that well:

First definition:

```python
# demo.Pair
float16 first
float16 second
```

```python
# demo.PairVector
demo.Pair[3] vector
```

```python
demo.PairVector pair_vector
```

Second definition:

```python
float16 first_0     # pair_vector.vector[0].first
float16 second_0    # pair_vector.vector[0].second
float16 first_1     # pair_vector.vector[1].first
float16 second_1    # pair_vector.vector[1].second
float16 first_2     # pair_vector.vector[2].first
float16 second_2    # pair_vector.vector[2].second
```

The code compatibility guarantee would partially defeat the advantages of the new version management logic.

-----

On the subject of binary compatibility: I've been silent lately because of this short statement you made in a different thread:

> 3.6.2.1 Valid encoded representation needs defining. One of the things im unsure of: Is a void field encoded to a non-zero value valid encoded?

I thought that I'm just going to throw in a few words to clarify things. However, what actually happened is that I opened a huge can of worms with a whole universe in it. I was sketching some bit patterns in my notepad as I received a notification about this issue; now let me go back to it.

kjetilkjeka, 2018-08-09:
> How do we express deprecation?

Let's address this concern before moving on to the bigger concerns.

> I have added a @deprecated keyword to the current draft a couple of weeks ago; it's supposed to be translated into language-specific deprecation markers, e.g. [[deprecated]] in C++ or DeprecationWarning in Python.

That's nice. I was playing around with a similar idea myself. What i was thinking about was adding 
`@lifecycle <pre-release|released|deprecated|removed>`, this way we can add a new type without releasing it, yank a buggy type and deprecate old types. This is how i vision it would work.

 - **Pre-realease** - do not generate code for this definition unless some dangerous keyword is presented.
 - **Released** - As normal (as it currently works)
 - **Deprecated** - Issue a warning that the type is deprecated and will be removed in the future.
 - **Removed** - Issue an error that this type has been through the deprecation process and is no longer usable.

Even if we initially only supported `deprecated` It would allow us to add other lifecycle items (like removed) later on without making a new directive.

pavel-kirienko, 2018-08-09:
I like the flexibility argument, but I have two questions:

1. What's the point of the `removed` option if the definition can be just physically removed to the same effect?
2. `pre-release` goes against the versioning guarantees that once something is published it can't be removed or changed without a proper deprecation process. If you want tentative volatile definitions, use version 0. I think we can make this work but that would be at the cost of extra complexity in the specification, so I am against it.

If the two options mentioned above are removed, we're back to just `deprecated` and `released`. I think the meaning of the keyword `@lifecycle` might be slightly less transparent than `@deprecated`, i.e. somebody who is not familiar with the UAVCAN specification might fail to understand what it means; `@deprecated` seems more approachable. Also, most published data type definitions will be in the `release` stage, meaning that it would make sense to assume `release` by default if not specified, so `@lifecycle release` becomes redundant.

I suggest keeping `@deprecated`.

kjetilkjeka, 2018-08-09:
I split the deprecation issue out to: #10 for making the issues easier to follow from the outisde

kjetilkjeka, 2018-08-09:
> What's the point of the removed option if the definition can be just physically removed to the same effect?

The main idea was to give information about potential DTID/name/path conflicts for future definitions. We should probably at least keep this element out of the mix until we can see what the dsdl checker ends up doing in a bit more detail.

> pre-release goes against the versioning guarantees that once something is published it can't be removed or changed without a proper deprecation process. If you want tentative volatile definitions, [...]

Well my intention was to not consider it "published" before it reached `released`. The `pre-release` was meant as a way for early testers to evaluate 2.0 definitions and the like before released.

It's actually a bit different than 0.x definitions for a few reasons:
 - It's quite more dangerous since it may conflict with released stable versions. This would need to make it *very hard*  to enable by mistake.
 - It should be thought more of as a way to develop the type replacing the current.
 - It can be used with "stable" types
 - It can even be used with 0.x types to 

---
My counter proposal would be to use `@lifecycle <released|deprecated>` with an implicit default to `@lifecycle deprecated`. This would allow most people not to think about it while allowing us to add more stages at a later point.

With that said I also totally accept `@deprecated`. We should in any case leave this issue open for a week or two in case someone else wants to bikeshed this further (Hey! perhaps `@deprecated` should be an alias for `@lifecycle release` :laughing: ). I won't object to you closing this thread with the decision you're in favor for when you feel it has cooked long enough, unless you want to discuss it further.

kjetilkjeka, 2018-08-09:
> I thought that I'm just going to throw in a few words to clarify things. However, what actually happened is that I opened a huge can of worms with a whole universe in it. I was sketching some bit patterns in my notepad as I received a notification about this issue; now let me go back to it.

Isn't the problem here essentially that a "valid encoding" and a "valid decoding" is two seperate things, and therefore defining a single "valid encoded representation" term is impossible (or at least needs to be the most liberal of these). I think if we change the definition to:

#### Bit compatibility
A structure definition `A` is *bit-compatible* with a structure definition `B` if both the following statements are true:
1. The encoding of any valid frame of type `B` can be decoded to a an element of type `A`.
2. For every value that can be decoded into an element of type `B` there exists (one or more) an element of type `A` that will be encoded into this value.

---

Or a bit more verbose.

Let's define `A` as the set of possible instantiations of a given dsdl definition.

Let's define `S(A)` as the bit_masks that respects ranges of dynamic structures when decoding DSDL. This means that if an element from `S(A)` was decoded into `A`: the total length, all array lengths and union discriminators would be valid. (this can be specified more constructively if needed, https://github.com/UAVCAN/uavcan.github.io/issues/40 can be used as a start if needed )

An encoding is a function from `A` to `S` (`e_A: A -> S(A)`). We do not require this function to be surjective but we do require it to be injective (two elements in `A` can only have the same encoding if they are equal). 

On the other hand, a decoding is a function from `S(A)` to `A` (d_A: S(A) -> A`). We do not require this function to be injective but we require it to be surjective (an element cannot exist without at least one valid encoding)

Let's define `B` as the set of possible instantiations for some other given dsdl definition.

For bit compatibility between the definitions that constructs `A` and `B` we require:
 - `S(A) = S(B)` (as in `==`) This is basically what i attempted to guarantee in https://github.com/UAVCAN/uavcan.github.io/issues/40
 - All representable elements in B is also representable in A: ![equation](https://latex.codecogs.com/gif.latex?%5Cforall%20b%20%5Cin%20B%2C%20%28%20d_B%20%5Ccirc%20e_A%20%5Ccirc%20d_A%20%5Ccirc%20e_B%29%28b%29%20%3D%20b)

pavel-kirienko, 2018-08-10:
> Or a bit more verbose.

So my approach was different - I was attempting to avoid defining the decoding function completely, expressing the relations in terms of the encoding function only: A is compatible with B if the set of all possible encoded representations of A is a superset of all possible encoded representations of B. However, void fields were giving me trouble, because they are by definition to be encoded as zero, which breaks the set relationship if at least one void bit is to be replaced with a non-void bit. I also tried defining an extended set of all possible encoded representations by allowing the void fields to take arbitrary values, but that created some ugly corner cases with dynamic fields (specifically, array length prefixes and union discriminators could no longer consume void bits because that broke the set relationship as well, making the types incompatible when they should be).

Your very last equation makes perfect sense. Let me try continuing with your approach.

pavel-kirienko, 2018-08-10:
Wait no, loss of information may occur at `d_A(e_B(b))` and `d_B(e_A(b))` - fields that are present in one definition may be missing in the other, so your last equation does not hold.

kjetilkjeka, 2018-08-10:
> Wait no, loss of information may occur at d_A(e_B(b)) and d_B(e_A(b)) - fields that are present in one definition may be missing in the other, so your last equation does not hold.

Since the definition is concerned with *`A` being bit compatible with `B`* loss of information may occur when going from `A` to `B` (since a void field may have been populated) but not the other way around.

First observe that `d_X(e_X(a)) = a` always holds (this is the identity function on the set `X`). But `e_ X(d_X(s)) = s` only holds when `s` is an element in the range of `e_X`

If we first define c as `c = e_ B(b)` we can then see that the equation `c = d_A(e_A(c))` always holds iff (if and only if) `c` is in the range of `e_A`.  This boils the second conditions down to the same as the range of `e_B` being a subset of `e_A`.

The consequence is that it's not backwards compatible to remove void fields (or generally remove information) but it is backwards compatible to populate void fields (or generally add information), i think this was what we wanted? Ìf you instead would like the relaxed definition where only the first condition are present. You're essentially asking for serialization compatibility as defined in https://github.com/UAVCAN/uavcan.github.io/issues/40

pavel-kirienko, 2018-08-13:
> The consequence is that it's not backwards compatible to remove void fields (or generally remove information) but it is backwards compatible to populate void fields (or generally add information), i think this was what we wanted?

That seems overly strict. One can imagine a use case where a new void field would need to be introduced, e.g. if a non-void field became deprecated or split into several smaller fields. I suggest staying generic; perhaps we should move this conversation back to https://github.com/UAVCAN/specification/pull/11 where I have proposed a very minimalistic formulation.

kjetilkjeka, 2018-08-21:
I think we need to talk about the concrete things we expect to have from compatibility data types and their concrete value:

## Problems

### Populating void fields
I think populating void fields is the prime example of what we're 100% sure that we want and doesn't cost us anything. The constraint is that newly populated fields must always be considered as an "optional" with a None variant of the default void serialization value. The following example is overly pedantic about this.

#### Example
```
# Version 1.0
uint8 foo
void8
```

```
# Version 1.1
uint8 foo

uint8 BAR_NONE = 0
uint8 bar
```

### Removing information
Any form of removing information must be forbidden. If this is not true we must expect that every important piece for information can be removed in the future. This is a destructive way to have to think when working with uavcan. And removing fields should always be semantically incompatible.
#### Example fields
```
# Version 1.0
uint8 a
uint8 b
```

```
# Version 1.1
uint8 a
void8
```

#### Examples consts
```
# Version 1.0
uint8 FOO = 42
uint8 bar
```

```
# Version 1.1
uint8 bar
```

### Unrolling arrays/structs or merging fields
Unrolling or enrolling structs and arrays and merging of fields can make the data types more convenient for the user, but it does not add or remove any information. The problem with this is that it also breaks code compatibility which is inconvenient for users. 

Instead of doing this on released data types we should experiment enough in the pre-releases that we "get it right". If we fail, we should still not do this for released datatypes *as the code breakage is much more severe and annoying than the benefit of being able to make these changes*.

#### Example unrolling/enrolling (stolen from Pavel)
```python
# demo.Pair
float16 first
float16 second
```

```python
# demo.PairVector
demo.Pair[3] vector
```

```python
demo.PairVector pair_vector
```

Second definition:

```python
float16 first_0     # pair_vector.vector[0].first
float16 second_0    # pair_vector.vector[0].second
float16 first_1     # pair_vector.vector[1].first
float16 second_1    # pair_vector.vector[1].second
float16 first_2     # pair_vector.vector[2].first
float16 second_2    # pair_vector.vector[2].second
```

#### Example merging fields (stolen from Pavel)
```python
# Alert flags are allocated at the bottom (from bit 0 upwards)
uint64 DC_UNDERVOLTAGE                       = 1 << 0
uint64 DC_OVERVOLTAGE                        = 1 << 1
uint64 DC_UNDERCURRENT                       = 1 << 2
uint64 DC_OVERCURRENT                        = 1 << 3
uint64 CPU_COLD                              = 1 << 4
uint64 CPU_OVERHEATING                       = 1 << 5
uint64 VSI_COLD                              = 1 << 6
uint64 VSI_OVERHEATING                       = 1 << 7
uint64 MOTOR_COLD                            = 1 << 8
uint64 MOTOR_OVERHEATING                     = 1 << 9
uint64 HARDWARE_LVPS_MALFUNCTION             = 1 << 10
uint64 HARDWARE_FAULT                        = 1 << 11
uint64 HARDWARE_OVERLOAD                     = 1 << 12
uint64 PHASE_CURRENT_MEASUREMENT_MALFUNCTION = 1 << 13

# Non-error flags are allocated at the top (from bit 63 downwards)
uint64 UAVCAN_NODE_UP                        = 1 << 56
uint64 CAN_DATA_LINK_UP                      = 1 << 57
uint64 USB_CONNECTED                         = 1 << 58
uint64 USB_POWER_SUPPLIED                    = 1 << 59
uint64 RCPWM_SIGNAL_DETECTED                 = 1 << 60
uint64 PHASE_CURRENT_AGC_HIGH_GAIN_SELECTED  = 1 << 61
uint64 VSI_MODULATING                        = 1 << 62
uint64 VSI_ENABLED                           = 1 << 63

# Status flags, see the constants above
uint64 flags
```

```python
# Alert flags are allocated at the bottom (from bit 0 upwards)
uint32 ALERT_DC_UNDERVOLTAGE                       = 1 << 0
uint32 ALERT_DC_OVERVOLTAGE                        = 1 << 1
uint32 ALERT_DC_UNDERCURRENT                       = 1 << 2
uint32 ALERT_DC_OVERCURRENT                        = 1 << 3
uint32 ALERT_CPU_COLD                              = 1 << 4
uint32 ALERT_CPU_OVERHEATING                       = 1 << 5
uint32 ALERT_VSI_COLD                              = 1 << 6
uint32 ALERT_VSI_OVERHEATING                       = 1 << 7
uint32 ALERT_MOTOR_COLD                            = 1 << 8
uint32 ALERT_MOTOR_OVERHEATING                     = 1 << 9
uint32 ALERT_HARDWARE_LVPS_MALFUNCTION             = 1 << 10
uint32 ALERT_HARDWARE_FAULT                        = 1 << 11
uint32 ALERT_HARDWARE_OVERLOAD                     = 1 << 12
uint32 ALERT_PHASE_CURRENT_MEASUREMENT_MALFUNCTION = 1 << 13
uint32 alert_flags

uint32 STATUS_UAVCAN_NODE_UP                       = 1 << 24
uint32 STATUS_CAN_DATA_LINK_UP                     = 1 << 25
uint32 STATUS_USB_CONNECTED                        = 1 << 26
uint32 STATUS_USB_POWER_SUPPLIED                   = 1 << 27
uint32 STATUS_RCPWM_SIGNAL_DETECTED                = 1 << 28
uint32 STATUS_PHASE_CURRENT_AGC_HIGH_GAIN_SELECTED = 1 << 29
uint32 STATUS_VSI_MODULATING                       = 1 << 30
uint32 STATUS_VSI_ENABLED                          = 1 << 31
uint32 status_flags
```

## Conclusion

If we were to break "code compatibility" it should be for the ability to add new features. I do not see enough value in mixing around with existing datatypes to justify doing so.

kjetilkjeka, 2018-08-21:
> That seems overly strict. One can imagine a use case where a new void field would need to be introduced, e.g. if a non-void field became deprecated or split into several smaller fields. I suggest [...]

What is the point of deprecating non-void fields? We cannot repopulate it in a sound way.

If every field can be removed you do not have much guarantee as a user of a data type either. We should not be able to remove features (anything) and still call something compatible.

pavel-kirienko, 2018-08-27:
I think it is important to ensure that our reasoning here is guided by the top-level design objectives and not by one's vision on how to implement the best diagnostic tools. ;)

It is crucial to ensure that we have all possible tools at our disposal when it comes to the ability to advance data type definitions while not breaking backward compatiblity with existing nodes, because updating fielded nodes is hard and expensive. The balance between ease of coding and ease of maintenance should be skewed towards the latter, which is why I argue that making code compatiblity mandatory is a step in a wrong direction.

> What is the point of deprecating non-void fields? We cannot repopulate it in a sound way.
> If every field can be removed you do not have much guarantee as a user of a data type either. We should not be able to remove features (anything) and still call something compatible.

This issue is supposed to be managed by the semantic compatiblity requirement. It should be possible to remove constants that ended up being unnecessary, or fields that turned out to be redundant. My example that you've provided in your earlier post here can be easily modified such that the existing wide integer field is split into two smaller fields with a void field in the middle - this is an example of a meaningful transition from a non-void field to a void field.

pavel-kirienko, 2018-12-11:
I am closing this as the discussion seems to have migrated to the forum completely:

- https://forum.uavcan.org/t/the-case-for-code-compatibility/194/
- https://forum.uavcan.org/t/simpler-and-better-dsdl-versioning/280/

closed 2018-12-11 by pavel-kirienko