OpenCyphal/specification #14: Continuous integration

pavel-kirienko, 2018-08-18:
We need to set up a CI to automatically build the spec doc and upload the resulting PDF somewhere. Ideally, for pull requests, the resulting PDF should be attached to a comment after every build. Travis supposedly can be configured to do that, see https://damien.pobel.fr/post/github-api-from-travisci/. Although I'm not sure if it's possible to upload an attachment using that API.

Would somebody like to volunteer?

kjetilkjeka, 2018-08-19:
Pavel, would it be possible to create a mail forward rule from spec-buildbot@uavcan.org (or similar) to my e-mail and other e-mail addresses you think is appropriate? Then I will register a new gh account with this e-mail.

pavel-kirienko, 2018-08-19:
Yes, it should be trivial, I'm going to have a look probably tomorrow or the day after.

pavel-kirienko, 2018-08-20:
Okay, you should have received an invitation to join a collaborative inbox. All mail sent to any address `@uavcan.org` will end up there. I suggest using the same address `maintainers@uavcan.org` everywhere for ease of maintenance.

kjetilkjeka, 2018-08-21:
> I suggest using the same address `maintainers@uavcan.org` everywhere for ease of maintenance.

I'm not sure if it's a good idea as we might get a few more gh logins in the future and they all need a unique e-mail address.

kjetilkjeka, 2018-09-24:
After some exploration of alternatives, I've come up with a new plan.

Gitlab CI/CD is great, I'm currently using it for a couple of personal projects. The most important improvement from Travis is the "build artifact" system that is essentially what we want.

I've tested out Gitlab CI for Github, it's not perfect, but it works better than travis would for this specific use case. I propose that we set up an uavcan gitlab organization so we're able to use the CI functionality.

You can see how gitlab creates an .pdf artifact from a github PR [here](https://github.com/kjetilkjeka/specification/pull/1)

I've already done the necessary setup on my fork (to test things out). Meaning it's trivial to set up this for the upstream repo as well. I can have this done by tomorrow. @pavel-kirienko 
 - Let me know if you have any objections to using Gitlab CI over Travis.
 - Let me know if I should (or you will) create an Gitlab organization for Uavcan.

As a bonus we will also get an automatic updated mirror of the github repo.

pavel-kirienko, 2018-09-24:
Great.

>  Let me know if you have any objections to using Gitlab CI over Travis.

No objections.

> Let me know if I should (or you will) create an Gitlab organization for Uavcan.

Please go ahead.

kjetilkjeka, 2018-09-24:
@pavel-kirienko @thirtytwobits let me know if you have any problems with this.

closed 2018-09-24 by kjetilkjeka