OpenCyphal/specification #20: Add a "changes from v0" section

thirtytwobits, 2018-09-06:
To help system architects that are evaluating a change from v0 to v1 of the protocol we should provide a section of the document that details the differences and describes strategies for migrating existing components.

pavel-kirienko, 2018-12-12:
As discussed at the call today, this should be a separate piece of documentation rather than a section in the specification. The main motivation for this separation is the expectation that this information will become obsolete very quickly, making it pointless to keep in the main specification document the life cycle of which is much longer.

I think it should be okay to describe the changes in a forum post, in the form of a migration guide. Would anyone like to volunteer?

MadRC, 2019-08-22:
Was this ever done in a forum post ? I’m trying to understand the main reasons for the changes and hardware implications.

pavel-kirienko, 2019-08-22:
This is not yet done on the object level. From your question I suspect that you are interested in a higher-level overview; for that please proceed here: https://forum.uavcan.org/t/uavcan-a-highly-dependable-publish-subscribe-protocol-for-real-time-intravehicular-networking/557

> hardware implications

No hardware implications for vendors. The transition from v1 to v0 can be done by updating the software only.

pavel-kirienko, 2020-03-11:
I expect this post should address it: https://forum.uavcan.org/t/uavcan-v1-crash-course/778

@MadRC if you could give it a look and perhaps provide feedback that would be awesome.

closed 2020-03-12 by pavel-kirienko