OpenCyphal/specification #41: Final proofreading - grammar, formatting, readability

pavel-kirienko, 2018-12-14:
@hamishwillee @mrpollo we are optimistically aiming to finalize the draft next week. What are our chances of getting assistance from Dronecode to get the document proofread and corrected?

hamishwillee, 2018-12-17:
Hi @pavel-kirienko ,

FYI I no longer directly work for Dronecode, but my management DIUx and Auterion are highly supportive of standards and protocol specification. So I can spend a couple of days on this.
A few points:
1. I find it challenging to do technical editing of specs in one go. If there are any bits of the spec that are known "good to go" then great if you could drip-feed to me. 
2. Xmas is coming. If this drags into next week things get complicated.
Feel free to slack-me with any additional info.

pavel-kirienko, 2018-12-17:
Okay, thanks, Hamish.

Right now I think nothing is good to go yet because after the missing chapters are finished (see https://github.com/UAVCAN/specification/issues/42) we will go over the doc again and possibly restructure stuff; proofreading probably can't be done until that is finished.

I should post here when we can start proofreading.

hamishwillee, 2018-12-17:
OK, please do. I'll do my best to review it in a timely manner once it is posted.

PR #68 (2020-01-21): Changeset To End All Changesets
Closes #65 "Review the data type compatibility rules" -- only semantic compatibility is left. The compatibility rules are drastically simplified. Instead of rigorous rules, the specification takes a high-level view, dictating that the integrator is responsible for the functional correctness. We will be able to introduce stricter rules later shall better solutions be found.

Closes #61 "Data type extensibility" -- we have a rudimentary polymorphism in place already.

Closes #54 "Replace must with shall".

Closes #41 "Final proofreading - grammar, formatting, read [... cut]
closed 2020-01-21 by pavel-kirienko