OpenCyphal/specification #86: Generate and publish HTML version of the specification

thirtytwobits, 2020-06-17:
The PDF is great but you can't provide links into it. We should also generate an HTML/web version from the same tex and publish to the uavcan.org domain.

pavel-kirienko, 2020-07-13:
FWIW page anchors seem to work, although I'm not sure if it's a standard feature:

https://helpx.adobe.com/acrobat/kb/link-html-pdf-page-acrobat.html

This link works for me in Firefox, opening the page 55: https://uavcan.org/specification/UAVCAN_Specification_v1.0-alpha.pdf#page=55. Does it work for you?

@hamishwillee Hamish, do you think we could realistically ask Auterion to kindly help us get our docs in order?

thirtytwobits, 2020-07-13:
It seems to work in Chrome. Nice.

hamishwillee, 2020-07-13:
@pavel-kirienko Maybe. I'll send you a message offline.

closed 2025-05-08 by pavel-kirienko