OpenCyphal/specification #98: Specify UAVCAN/UDP

pavel-kirienko, 2020-09-12:
The draft is available at https://pyuavcan.readthedocs.io/en/latest/.apidoc_generated/pyuavcan.transport.udp.html

The discussion is at https://forum.uavcan.org/t/alternative-transport-protocols-in-uavcan/324

ASMfreaK, 2022-08-09:
Please cyphalize the link: https://pycyphal.readthedocs.io/en/stable/api/pycyphal.transport.udp.html#cyphal-udp-transport-overview

pavel-kirienko, 2023-02-03:
As discussed at the call, Cyphal/UDP will be available in v1.0 only as an experimental capability with the possibility of breaking changes in the future. The most recent draft is available on the forum here: https://forum.opencyphal.org/t/cyphal-udp-architectural-issues-caused-by-the-dependency-between-the-nodes-ip-address-and-its-identity/1765/60?u=pavel.kirienko

PR #127 (2023-05-10): Specify Cyphal/UDP [experimental]
- Add Cyphal/UDP specification (marked as experimental); fixes #98.
- Move CRC implementations to the appendix.

The specification is quite concise because the key concepts are already covered in section "4.1 Abstract concepts". In the new section, I only had to cover the layouts.
commit c6ed43db (2023-05-10): Specify Cyphal/UDP [experimental] (#127)

- Add Cyphal/UDP specification (marked as experimental); fixes #98.
- Move CRC implementations to the appendix.

The specification is quite concise because the key concepts are already
covered in section "4.1 Abstract concepts". In the new section, I only
had to cover the layouts.
closed 2023-05-10 by pavel-kirienko